crysxd/OctoApp-Plugin #2: PSU Control stops working

arut16, 2021-10-30:
**Describe the bug**
PSU Control Stops working after installation of the plugin.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the plugin
2. Try to turn on/off the printer

**Expected behavior**
Sending notifications and not interfere with other plugins

**Your software and hardware (please complete the following information):**
 - OctoPrint or OctoPi version: 

Octoprint 1.7.0 Python 3.7.3 OctoPi 0.17.0

 - Browser [e.g. chrome, safari] : Chrome Version 95.0.4638.54
 - Plugin version [e.g. 0.1.0] : 1.0.0

crysxd, 2021-10-31:
This is pretty weird, I have the exact same setup here.

In the ticket for OctoApp you said that the PSU is turned on and then off again after a few seconds. Is this still the case?

arut16, 2021-10-31:
Actually, it's kinda weird. Sometimes, then I trigger the power button, the relay just close and then open after few seconds (1-3 sec). Sometimes, it just immediately close and open again. And the last time, it doesn't even react by punching the power button...
But after deactivating the plugin, restarting octoprint, all works again.

crysxd, 2021-10-31:
Could you add OctoApp and PSUControl with debug logging level and send me the octoprint.log after the issue happened again? (in the settings > logging) Maybe it's best to delete the octoprint.log as well when you enable the debug logging so we only have the relevant logs 

[image]

I will also start a discussion in the OctoPrint Discord, maybe somebody knows what's up here. Looking at the source code of both plugins, I'm not sure how they'd interfere

closed 2021-11-21 by crysxd