crysxd/OctoApp-Plugin #7: Clean up App Notifications

escfoe2, 2022-01-03:
[image]

I feel like these can be combined into one notification, instead of separate ones.  I also notice that it always says 99% after the print has completed, when it should say 100%.  Or just Done.  Those are pet peeves of mine.  Many people already have several other apps that have push notifications.  Getting 3 different notifications from the same app is a bit annoying.

crysxd, 2022-01-06:
The 100% notification is not send but instead the app moves from 99% to done. If that's not the case for you we can look into why this does not work.

I understand you want old "completed" and "filament required" notifications to be cleaned automatically? Completed notifications should be easy to do, when the next print starts we can remove old ones. Filament change...not sure if it's as easy because I'm not sure about the event that would trigger the cleanup. Most likely that would be when the print continues all filament notifications are removed...

escfoe2, 2022-01-06:
What I'm asking is to reuse the existing notification that's active during the print.  The one that shows the current print percentage.  Just update that notification with relevant data, such as print paused, or complete.  Instead of generating a new one.

crysxd, 2022-01-07:
I don't think this is possible because the different notifications have different notifications categories. This allows users to change how notifications are displayed, e.g. on the Lock Screen or not and is used a lot. I can test if the notification for the progress can change its category, but I don't think this will correctly work on all Android phones. I also can't remove and post the progress notification again as it's linked to a service, removing the notification to post it again with a different category will require me to restart the service and Android might not allow me to start it again once it's stopped.

I can integrate the cleanup which will already remove all old notifications for you, meaning you have max 2 notifications at a time when the filament change is active and only one notification in any other situation (given that the 99% -> done switch works correctly for you?)

escfoe2, 2022-01-07:
The 99% to Done does not work for me.  It just sticks at 99%.

I don't understand why they can't all use the same notification category.  And if categories really are needed, they can be handled within the app itself by toggling a particular notification type and handling that toggle within the app.  Having a separate notification may be nice for some, and keeping it an option isn't an issue, but I believe it would make more sense to unify them into one.

Plenty of apps use a single notification and update it as needed for different events.  There's definitely a way.

crysxd, 2022-01-07:
Can you send me a bug report from within the app after the 99% notification should have been removed and the "print done" notification showed up? You can do so from the app's main menu > Settings > FAQ & help > Report bug. This should happen within a few minutes after the event as the app only keeps limited logs cached.

Other users want to have different notification categories, e.g. they want to see the progress notification only in their notification shade and not on the Lock Screen but they want the filament change notification to be visible on the Lock Screen as well. This can't be handled internally. I can make an option, but I don't see the need tbh. If the app removes previous notifications correctly (which is not the case for you), there is little to no difference between a cleanup and technically using the same notification.

closed 2022-02-11 by crysxd