crysxd/OctoApp-Plugin #9: Push notification when bed has preheated for XX minutes

bcat, 2022-01-08:
**Is your feature request related to a problem? Please describe.**
I like to preheat my printing bed for a few minutes before starting a print to ensure consistent first-layer adhesion. Right now I kick off the preheat, then manually check the OctoPrint Web UI to see how many minutes have elapsed.

**Describe the solution you'd like**
It'd be really nice to have OctoApp send an (optional) push notification after the bed has been preheated to temperature for a configurable number of minutes.

**Describe alternatives you've considered**
Not preheating (no plans to change that), manually monitoring time/temperature (what I do now, but it's a bit of a pain).

**Additional context**
The [OctoPod iOS app](https://github.com/gdombiak/OctoPrint-OctoPod) has a similar feature, which is why I got the idea for this.

crysxd, 2022-01-10:
So basically you want to get an reminder when the print bed preheated for XX minutes? This would count from the time you set the target temperature?

bcat, 2022-04-10:
Yeah, that's basically it. Otherwise, I tended to forget the bed was preheated.

FWIW, since requesting this I've switched from PEI on glass to PEI on spring steel, and preheating matters much less (since the surface of the bed is at target temperature almost immediately after the bed sensor is). But I can still see this being useful for folks with glass beds.

crysxd, 2024-10-12:
You can send custom notifications with this command in the OctoPrint terminal or from a Gcode file
`OCTOAPP_NOTIFY MESSAGE=Test 1234`

Using this you can create a custom control in OctoPrint:
https://docs.octoprint.org/en/master/features/custom_controls.html

The custom control can ask you for a temperature and then run the following gcode commands:

```
M190 R%(temp)
OCTOAPP_NOTIFY MESSAGE=Bed is now at %(temp)º
```

closed 2024-10-12 by crysxd