crysxd/OctoApp-Plugin #10: FeatureRequest: Ophom (Octoprint Philips Hue Outlet Manager) Support

Sakrapher, 2022-02-21:
Hi there,
I would like the support of the Ophom plugin.
Ophom can control ZigBee plugs directly via Hue bridge. 

https://plugins.octoprint.org/plugins/ophom/
https://github.com/Salamafet/ophom

Best Regards

crysxd, 2022-02-23:
I don't see any option in the plugin to manually control the plugs, thus there is also not really anything I can integrate? There is OctoHue which I plan on integrating soon which allows you to toggle a light

Sakrapher, 2022-03-05:
OctoHue just is for controlling lights, ophom is a plugin like PSU-Control.

crysxd, 2022-03-10:
Ophom has no manual controls, it's all automated (tell me if I'm wrong), so there is nothing I can integrate with

closed 2022-03-10 by crysxd