crysxd/OctoApp-Plugin #11: Newest plugin problems

FransBothma, 2022-05-02:
Newest plugin gives errors and does not install.

crysxd, 2022-05-03:
Could you be a bit more specific? What error is shown and where do you run OctoPrint on? Ideally, please provide the [system info bundle](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) here

FransBothma, 2022-05-03:
Hi, I am running octoprint on an android phone using Octo4A, phone is a Hauwei P30 Pro. Connected to an Ender 3v2. 

System Info:

browser.user_agent: Mozilla/5.0 (Linux; Android 12; 2107113SG) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/101.0.4951.41 Mobile Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 8
env.hardware.freq: 2146.25
env.hardware.ram: 7976067072
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.python.pip: 20.3.4
env.python.version: 3.9.5
env.python.virtualenv: False
octoprint.safe_mode: False
octoprint.version: 1.7.3
systeminfo.generator: zipapi

FransBothma, 2022-05-03:
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3uhzym3u/pillow_47f192e013894303a4ffa91b8306b757/setup.py'"'"'; __file__='"'"'/tmp/pip-install-3uhzym3u/pillow_47f192e013894303a4ffa91b8306b757/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-h6ug90m3/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/pillow Check the logs for full command output.
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
Close

crysxd, 2022-05-03:
Thanks! Can you please provide the `plugin_softwareupdate_console.log` file as well? You can get it in the OctoPrint settings under "Logging". Sorry for the many questions :)

FransBothma, 2022-05-03:
2022-04-29 15:31:04,301   /usr/bin/python3 -m pip --disable-pip-version-check install https://github.com/crysxd/OctoApp-Plugin/archive/1.0.10.zip --no-cache-dir
2022-04-29 15:31:24,069 > Collecting https://github.com/crysxd/OctoApp-Plugin/archive/1.0.10.zip
2022-04-29 15:31:32,112 > Downloading https://github.com/crysxd/OctoApp-Plugin/archive/1.0.10.zip (2.8 MB)
2022-04-29 15:31:51,453 > Requirement already satisfied: OctoPrint in /usr/lib/python3.9/site-packages (from OctoApp==1.0.10) (1.7.3)
2022-04-29 15:31:51,455 > Requirement already satisfied: pycryptodome==3.4.3 in /usr/lib/python3.9/site-packages (from OctoApp==1.0.10) (3.4.3)
2022-04-29 15:31:59,486 > Collecting pillow>=8.4.0
2022-04-29 15:32:00,494 > Downloading Pillow-9.1.0.tar.gz (49.8 MB)
2022-04-29 15:34:14,904 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2021.2.23)
2022-04-29 15:34:14,909 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2021.10.11)
2022-04-29 15:34:14,910 > Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2022.3.28)
2022-04-29 15:34:14,911 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.1.1)
2022-04-29 15:34:14,912 > Requirement already satisfied: markdown<3.2,>=3.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (3.1.1)
2022-04-29 15:34:14,914 > Requirement already satisfied: wrapt<1.13,>=1.12.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.12.1)
2022-04-29 15:34:14,916 > Requirement already satisfied: flask<2,>=1.1.4 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.1.4)
2022-04-29 15:34:14,918 > Requirement already satisfied: Jinja2<3,>=2.11.3 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2.11.3)
2022-04-29 15:34:14,921 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.5.0)
2022-04-29 15:34:14,921 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.0.0)
2022-04-29 15:34:14,922 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2.0)
2022-04-29 15:34:14,923 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.0.1)
2022-04-29 15:34:14,923 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.1.0)
2022-04-29 15:34:14,924 > Requirement already satisfied: cachelib<0.2,>=0.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.1.1)
2022-04-29 15:34:14,925 > Requirement already satisfied: PyYAML<6,>=5.4.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (5.4.1)
2022-04-29 15:34:14,926 > Requirement already satisfied: pyserial<4,>=3.4 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (3.5)
2022-04-29 15:34:14,926 > Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.8.0)
2022-04-29 15:34:14,927 > Requirement already satisfied: watchdog==0.10.4 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.10.4)
2022-04-29 15:34:14,927 > Requirement already satisfied: sarge==0.1.6 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.1.6)
2022-04-29 15:34:14,928 > Requirement already satisfied: netifaces<1,>=0.11 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.11.0)
2022-04-29 15:34:14,929 > Requirement already satisfied: pylru<2,>=1.2 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.2.1)
2022-04-29 15:34:14,929 > Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.8.2)
2022-04-29 15:34:14,930 > Requirement already satisfied: requests<3,>=2.26.0 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2.27.1)
2022-04-29 15:34:14,930 > Requirement already satisfied: semantic_version<3,>=2.8.5 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (2.9.0)
2022-04-29 15:34:14,931 > Requirement already satisfied: psutil<6,>=5.8 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (5.8.0)
2022-04-29 15:34:14,932 > Requirement already satisfied: Click<8,>=7.1.2 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (7.1.2)
2022-04-29 15:34:14,932 > Requirement already satisfied: future<1,>=0.18.2 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.18.2)
2022-04-29 15:34:14,933 > Requirement already satisfied: websocket-client<1,>=0.59 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (0.59.0)
2022-04-29 15:34:14,934 > Requirement already satisfied: emoji<2,>=1.4.2 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.7.0)
2022-04-29 15:34:14,934 > Requirement already satisfied: sentry-sdk<2,>=1.3.1 in /usr/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.0.10) (1.5.8)
[... cut ...] ull command output.
2022-05-02 20:51:41,093 > Running setup.py install for pillow: finished with status 'error'
2022-05-02 20:51:41,094 > Rolling back uninstall of Pillow
2022-05-02 20:51:41,094 > Moving to /usr/lib/python3.9/site-packages/PIL
2022-05-02 20:51:41,095 > from /usr/lib/python3.9/site-packages/~IL
2022-05-02 20:51:41,095 > Moving to /usr/lib/python3.9/site-packages/Pillow-8.2.0-py3.9.egg-info
2022-05-02 20:51:41,096 > from /usr/lib/python3.9/site-packages/~illow-8.2.0-py3.9.egg-info

FransBothma, 2022-05-03:
> Thanks! Can you please provide the `plugin_softwareupdate_console.log` file as well? You can get it in the OctoPrint settings under "Logging". Sorry for the many questions :)

It's no problem at all !! Thank you for the support !!
If you don't ask, I won't know what you need. Hehe

crysxd, 2022-05-03:
Made a small change, can you try installing the plugin manually via this URL:
`https://github.com/crysxd/OctoApp-Plugin/archive/release.zip`

You can do so in the OctoPrint settings > Plugin Manager > Ger more > ...from URL

FransBothma, 2022-05-03:
Thank you! I will have to wait untill after work sadly.

FransBothma, 2022-05-03:
Great news !! It installed successfully and works !!!!

crysxd, 2022-05-04:
Cool! I think the actual issue has to do with octo4a though. I checked with the other OctoPrint devs and the solution would have been to "install plugin extras" in octo4a 🙃

[image]

closed 2022-05-04 by crysxd

FransBothma, 2022-05-04:
Wierdly that is one of the setup steps for Octo4A, to install the plugin extras, so mine was installed. But will have a look at it if I get any other problems. Thank you for your help !!