crysxd/OctoApp-Plugin #13: Installation issue 1.0.12

MrPota2, 2022-08-08:
Unable to use Octoapp. Says it is installed successfully but does not show up on restart. Running octoprint on windows 10.

[systeminfo.txt](https://github.com/crysxd/OctoApp-Plugin/files/9284508/systeminfo.txt)
[terminal.txt](https://github.com/crysxd/OctoApp-Plugin/files/9284509/terminal.txt)
[octoprint.log](https://github.com/crysxd/OctoApp-Plugin/files/9284510/octoprint.log)
[serial.log](https://github.com/crysxd/OctoApp-Plugin/files/9284511/serial.log)
[plugin_pluginmanager_console.log](https://github.com/crysxd/OctoApp-Plugin/files/9284512/plugin_pluginmanager_console.log)
[plugin_softwareupdate_console.log](https://github.com/crysxd/OctoApp-Plugin/files/9284513/plugin_softwareupdate_console.log)
[plugin_softwareupdate_update.log](https://github.com/crysxd/OctoApp-Plugin/files/9284514/plugin_softwareupdate_update.log)

crysxd, 2022-08-23:
Hi!

Somehow missed this issue, will look into it.

crysxd, 2022-09-18:
Closing due to inactivity

closed 2022-09-18 by crysxd