crysxd/OctoApp-Plugin #14: Notifications for configurable layer lines

jbienz, 2022-08-30:
**Is your feature request related to a problem? Please describe.**
Feature request.

**Describe the solution you'd like**
I would really like to be able to configure OctoApp to deliver a push notification on my phone when the print is about to reach specific layer lines. I would like to check on the print when certain features start, for example bridging gaps. This would allow me to cancel the print if I've made a mistake in print planning.

**Describe alternatives you've considered**
I've looked for another option but the only ones I could find are OctoPod and Telegram. I don't use Telegram and I have an Android device.

**Additional context**
None, but thank you for considering!

crysxd, 2022-08-30:
This is already planned, will close as duplicate to https://gitlab.com/realoctoapp/octoapp/-/issues/1648

closed 2022-08-30 by crysxd

crysxd, 2022-08-30:
Quickly checked my planning, without making any promises I will try to add this to 1.17 for around November