crysxd/OctoApp-Plugin #17: Cannot install the plugin due to "pycryptodome"

ramifara, 2022-12-05:
when trying to install the plugin i get this error :

Installing collected packages: pycryptodome, OctoApp
Running setup.py install for pycryptodome: started
ERROR: Command errored out with exit status 1:
command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rlveydxy/pycryptodome_91144e1008914472b0d0876940165ecd/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rlveydxy/pycryptodome_91144e1008914472b0d0876940165ecd/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-dv_92pho/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/pycryptodome
cwd: /tmp/pip-install-rlveydxy/pycryptodome_91144e1008914472b0d0876940165ecd/
Complete output (335 lines):
Testing support for clang
Target does not support clang
Testing support for gcc
Target does not support gcc
Testing support for stdint.h header
Target does not support stdint.h header
Testing support for 128-bit integer
Target does not support 128-bit integer
Testing support for cpuid.h header
Target does not support cpuid.h header
Testing support for intrin.h header
Target does not support intrin.h header
Testing support for posix_memalign
Target does not support posix_memalign
Testing support for memalign
Target does not support memalign
Testing support for SSE2(intrin.h)
Target does not support SSE2(intrin.h)
Testing support for SSE2(x86intrin.h)
Target does not support SSE2(x86intrin.h)
Testing support for SSE2(emmintrin.h)
Target does not support SSE2(emmintrin.h)
Warning: compiler does not support AESNI instructions
Warning: compiler does not support CLMUL instructions
running install
running build
running build_py
creating build/lib.linux-aarch64-3.9
creating build/lib.linux-aarch64-3.9/Crypto
copying lib/Crypto/__init__.py -> build/lib.linux-aarch64-3.9/Crypto
creating build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_ocb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_cbc.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/__init__.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/Blowfish.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/ARC4.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_cfb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/PKCS1_OAEP.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/ChaCha20.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_eax.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/CAST.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_openpgp.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/PKCS1_v1_5.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_siv.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_ctr.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_ecb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_EKSBlowfish.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/DES3.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/DES.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/ARC2.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_gcm.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_ofb.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/Salsa20.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/ChaCha20_Poly1305.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/AES.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
copying lib/Crypto/Cipher/_mode_ccm.py -> build/lib.linux-aarch64-3.9/Crypto/Cipher
creating build/lib.linux-aarch64-3.9/Crypto/Hash
copying lib/Crypto/Hash/MD2.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
copying lib/Crypto/Hash/SHAKE128.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
copying lib/Crypto/Hash/SHA384.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
copying lib/Crypto/Hash/keccak.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
copying lib/Crypto/Hash/CMAC.py -> build/lib.linux-aarch64-3.9/Crypto/Hash
[... 259 lines cut ...]
copying lib/Crypto/Math/Numbers.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
copying lib/Crypto/Math/_IntegerGMP.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
copying lib/Crypto/Math/_IntegerBase.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
copying lib/Crypto/Math/Primality.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
copying lib/Crypto/Math/_IntegerNative.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
copying lib/Crypto/Math/_IntegerCustom.pyi -> build/lib.linux-aarch64-3.9/Crypto/Math
running build_ext
building 'Crypto.Hash._MD2' extension
creating build/temp.linux-aarch64-3.9
creating build/temp.linux-aarch64-3.9/src
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=64 -DLTC_NO_ASM -Isrc/ -I/usr/include/python3.9 -c src/MD2.c -o build/temp.linux-aarch64-3.9/src/MD2.o
gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
error: command '/usr/bin/gcc' failed with exit code 1
----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rlveydxy/pycryptodome_91144e1008914472b0d0876940165ecd/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rlveydxy/pycryptodome_91144e1008914472b0d0876940165ecd/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-dv_92pho/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/pycryptodome Check the logs for full command output.
Running setup.py install for pycryptodome: finished with status 'error'
Looks like the plugin was already installed. Forcing a reinstall.
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpaoc5e4bv/OctoApp-Plugin-release.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Processing /tmp/tmpaoc5e4bv/OctoApp-Plugin-release.zip
Using legacy 'setup.py install' for OctoApp, since package 'wheel' is not installed.
Installing collected packages: OctoApp
Running setup.py install for OctoApp: started
Running setup.py install for OctoApp: finished with status 'done'
Successfully installed OctoApp-1.1.0

crysxd, 2022-12-05:
`Successfully installed OctoApp-1.1.0`
What's the problem?

crysxd, 2023-01-19:
Closing due to inactivity

closed 2023-01-19 by crysxd