crysxd/OctoApp-Plugin #20: [Feature Request] Increase the layer height resolution to two decimal places in the Android app Gcode Preview screen

CoryCharlton, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
When I'm printing layer heights that have a resolution of two decimal places (ex: 0.12mm) the layer height is rounded when viewed in the Android app

**Describe the solution you'd like**
Increase the layer height resolution to two decimal places in the Android app Gcode Preview screen

**Describe alternatives you've considered**
Getting my treatment for my OCD and not worrying about trivial things such as this

**Additional context**
Here's a comparison of a print I have going on right now with a 0.12mm layer height:

[image]

[image]

crysxd, 2023-02-04:
My first thought was that this is a small bug, but turns out I actually put a bit of thought into it. If you show it with the "true" precision, e.g. 0.84, then the numbers will change length as you scrub through the layers using the sliders whenever you hit a height that has only one fraction digit, which looks not very nice. 

Simple solution: Now it's always two fraction digits: `1.00 mm` `0.80 mm` `0.84mm`. Hope nobody complains it's too precise now :D

Will not make a dedicated update for this small change, but will be flushed out next week I guess

Thanks for the feedback!

closed 2023-02-04 by crysxd

CoryCharlton, 2023-02-04:
Thanks so much for taking my feedback into consideration.

> Hope nobody complains it's too precise now :D

As an Android developer with several applications in the Play Store I know how challenging it can be to make everyone happy :D

Speaking of which I must say that OctoApp is a fantastic app. Very beautiful and well designed. Keep up the great work 👍

crysxd, 2023-02-04:
Thank you! Happy you like it!