crysxd/OctoApp-Plugin #29: Webcam URL calculation in app

gjedeer, 2023-05-03:
I have Octoprint exposed behind nginx like https://domain/octoprint/. This worked great with Octoapp except for the webcam. It seems to have a hard coded webcam URL to https://domain/webcam/?action=stream - ignoring the path part of the URL.

Would it be possible to add the path part to the generated URL (or if for some reason it's impossible, add a separate setting to webcam URL?)

crysxd, 2023-05-03:
Please report a bug via the apps FAQ and help menu so I can see the app logs and what the app is doing. What you describe should work.

Thanks!

closed 2023-05-03 by crysxd

crysxd, 2023-05-03:
Just a hunch, if your path starts with a `/` the app will behave like you describe. Make sure it's actually a relative path, e.g. `webcam/?action=stream` and not `/webcam/?action=stream`

crysxd, 2023-05-04:
Thanks for the update! Yes, you are correct where the path is coming from. Just to clarify, with the base URL `https://domain/octoprint/` as in your example, the absolute path `/octoprint/webcam?action=stream` should be equivalent to `webcam?action=stream`. The absolute path is appended to the host only (`https://domain` + `/octoprint/webcam?action=stream`) and the relative path on the full base URL (`https://domain/octoprint/` + `webcam?action=stream`)