crysxd/OctoApp-Plugin #39: I can't install OctoApp Plugin

phiphila, 2023-10-02:
## Please do not open tickets with bugs. Use the app's bug report feature instead (main menu > FAQ & help > report a bug)
I understood but it more simply to post pictures and code here

when I try to install your plugin on octoprint taht don't work correctly.

here is the install log:

`Installing plugin "OctoApp" from https://github.com/crysxd/OctoApp-Plugin/archive/refs/heads/release.zip...
/home/Octopi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmp_ojft0dr/OctoApp-Plugin-release.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp_ojft0dr/OctoApp-Plugin-release.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoApp==1.3.2) (1.9.2)
Requirement already satisfied: pillow in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoApp==1.3.2) (9.5.0)
Requirement already satisfied: pycryptodome>=3.15.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoApp==1.3.2) (3.19.0)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2021.10.11)
Requirement already satisfied: requests<3,>=2.28.2 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.31.0)
Requirement already satisfied: werkzeug<2.3,>=2.2.3 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.2.3)
Requirement already satisfied: markdown<4,>=3.4.3 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (3.4.3)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (21.3.0)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.12.1)
Requirement already satisfied: sarge==0.1.7.post1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.1.7.post1)
Requirement already satisfied: websocket-client<2,>=1.5.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.5.2)
Requirement already satisfied: wrapt<1.16,>=1.15 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.15.0)
Requirement already satisfied: sentry-sdk<2,>=1.19.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.24.0)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2021.2.23)
Requirement already satisfied: pyserial<4,>=3.5 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (3.5)
Requirement already satisfied: Flask-Login<0.7,>=0.6.2 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.6.2)
Requirement already satisfied: netifaces<1,>=0.11 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.11.0)
Requirement already satisfied: colorlog<7,>=6.7.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (6.7.0)
Requirement already satisfied: tornado<7,>=6.2 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (6.2)
Requirement already satisfied: pydantic<2,>=1.10.7 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.10.8)
Requirement already satisfied: zeroconf==0.39.4 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.39.4)
Requirement already satisfied: passlib<2,>=1.7.4 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.7.4)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.8.0)
Requirement already satisfied: PyYAML<7,>=6.0.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (6.0.1)
Requirement already satisfied: pkginfo<2,>=1.9.6 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.9.6)
Requirement already satisfied: semantic-version<3,>=2.10.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.10.0)
Requirement already satisfied: regex in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2022.9.13)
Requirement already satisfied: Flask-Limiter<4,>=3.3.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (3.3.1)
Requirement already satisfied: wheel in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.34.2)
Requirement already satisfied: emoji<3,>=2.2.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.2.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.0)
Requirement already satisfied: flask<2.3,>=2.2.3 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.2.5)
Requirement already satisfied: pylru<2,>=1.2.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.2.1)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (3.1.0)
Requirement already satisfied: feedparser<7,>=6.0.10 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (6.0.10)
Requirement already satisfied: Click<9,>=8.1.3 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (8.1.3)
Requirement already satisfied: watchdog<3,>=2.3.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.3.1)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.2.6)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.5.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.5.0)
Requirement already satisfied: OctoPrint-PiSupport>=2023.5.24 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2023.7.31)
Requirement already satisfied: frozendict<3,>=2.3.7 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.3.8)
Requirement already satisfied: pathvalidate<3,>=2.5.2 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (2.5.2)
Requirement already satisfied: future<1,>=0.18.3 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.18.3)
Requirement already satisfied: blinker<2,>=1.6.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.6.2)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.2.0)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (0.10.2)
Requirement already satisfied: psutil<6,>=5.9.4 in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (5.9.5)
Requirement already satisfied: unidecode in /home/Octopi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoApp==1.3.2) (1.3.6)
Requirement already satisfied: ifaddr>=0.1.7 in /home/Octopi/oprint/lib/python3.9/site-packages (from zeroconf==0.39.4->OctoPrint->OctoApp==1.3.2) (0.2.0)
Requirement already satisfied: async-timeout>=4.0.1 in /home/Octopi/oprint/lib/python3.9/site-packages (from zeroconf==0.39.4->OctoPrint->OctoApp==1.3.2) (4.0.2)
Requirement already satisfied: argon2-cffi-bindings in /home/Octopi/oprint/lib/python3.9/site-packages (from argon2-cffi<22,>=21.3.0->OctoPrint->OctoApp==1.3.2) (21.2.0)
Requirement already satisfied: more-itertools>=5.0.0 in /home/Octopi/oprint/lib/python3.9/site-packages (from class-doc<0.3,>=0.2.6->OctoPrint->OctoApp==1.3.2) (9.1.0)
Requirement already satisfied: sgmllib3k in /home/Octopi/oprint/lib/python3.9/site-packages (from feedparser<7,>=6.0.10->OctoPrint->OctoApp==1.3.2) (1.0.0)
[... cut ...]
Stored in directory: /tmp/pip-ephem-wheel-cache-xspvchj9/wheels/b2/0b/a0/2c5fb481dc301e3fd0373624daeec95baf333a6e6c55870cc9
Successfully built OctoApp
Installing collected packages: OctoApp
Attempting uninstall: OctoApp
Found existing installation: OctoApp 1.3.2
Uninstalling OctoApp-1.3.2:
Successfully uninstalled OctoApp-1.3.2
Successfully installed OctoApp-1.3.2
Done!`

and the picture after the installation:
[image]

after a restart, your plugin is not visible in the list of plugins and is not detected/available on my iphone because it offers me to install this plugin.

crysxd, 2023-10-02:
Pleas edit your comment and use ``` for the code. It's all one line, impossible to read

crysxd, 2023-10-02:
Also please send an email to hello@octoapp.eu, we can continue there. As you read I do not offer support here as I need to streamline my support process as much as possible. Will close here and wait for your email. Please still update the comment so I can read the logs properly.

closed 2023-10-02 by crysxd