crysxd/OctoApp-Plugin #44: Support for PrusaMMU

wiseguy56, 2023-12-19:
**Is your feature request related to a problem? Please describe.**
Was using PrusaMMU to select filament for the MMU before using octoapp. Opening the web interface to select a filament was very clunky. Support for Mmu2filamentselect is very nice but miss having the features of PrusaMMU. 

**Describe the solution you'd like**
Add plugin support for PrusaMMU plugin. 

**Describe alternatives you've considered**
Swapped to using Mmu2filamentselect but the features definitely leave something to be desired coming from PrusaMMU. The current selected filament is somewhat useful/fun but error notifications or current state through octoapp with the new MMU 3.0.0 update would be awesome since the MMU can get stuck frequently, along with the ability to remap tools. 
PrusaMMU is also actively being updated.

crysxd, 2023-12-19:
This is tracked here: https://gitlab.com/realoctoapp/octoapp/-/issues/2411

I'll try to work on it in January, it's generally quite similar so I don't expect it to be too much work. My main focus was Klipper in the last months, so this had to wait :)

closed 2023-12-19 by crysxd

wiseguy56, 2023-12-19:
Sweet, thanks!