crysxd/OctoApp-Plugin #60: OctoApp and Klipper

Daybreak360, 2024-02-24:
Sorry for opening this in github, I am having trouble with the ios mail app. 

I having difficulty adding the companion app to my Creality K1C. 

Attached is the error i am receiving when I am trying to install it via the instructions given in the app. I have installed this successfully on two other klipper machines and was hoping I could get this one working as well. Thanks for the amazing app! 
[image]

crysxd, 2024-02-25:
Where did you find the instructions to install the plugin? :)

This is a screenshot from the app, did you find them anywhere else? Might have forgot to update something
[image]

Please run the above commands!

- `git clone https://github.com/crysxd/OctoApp-Plugin octoapp`
- `cd octoapp`
- `./install.sh`

Daybreak360, 2024-02-25:
I got it from the iOS app, but I swear that wasn't there in my app. Thank you so much for sharing!!

crysxd, 2024-02-25:
I added the second part last week, maybe your app didn't fetch the new content yet! :) 
Let me change the instructions to `sh install.sh`, somebody else mentioned the same!

crysxd, 2024-02-25:
By the way, can you check if `wget https://octoapp.eu/install.sh` works? `/bin/bash` doesn't exist for you as well, but we can use `sh` instead. If `wget` works my shorthand command can be used again :D Still have to add it to KIAUH...

Daybreak360, 2024-02-25:
I uninstalled and deleted the octoapp directory to try the wget. This is what I received:

root@K1C-6448 /root [#] wget https://octoapp.eu/install.sh
Connecting to octoapp.eu (151.101.65.195:443)
wget: note: TLS certificate validation not implemented
wget: can't open 'install.sh': File exists

I have real basic knowledge of it all so I am not totally sure of how this is all supposed to work. I know I have an install.sh in my ~ folder from KIAUH. So I created an new dir and tried the command from there, it worked and downloaded the install.sh to the new folder but it wouldn't actually run, I am guessing it is missing other dependancies like the ones that are downloaded when you do the git clone. If you were just checking if wget works then, yes it did! It totally downloaded in a clean dir with no install.sh already there.

I hope that helps!