crysxd/OctoApp-Plugin #64: Change dialog prompt order when Spool Manager and PrusaMMU are present

Java-SE-Binary, 2024-03-30:
When starting a print from the app with Spool Manager AND PrusaMMU plugins installed, the app will first ask to select which material to use from Spool Manager, then once that is selected, the dialog for PrusaMMU will appear, asking which tool to use. The problem with this, selecting a material will automatically select it for Tool 0, and swap the spool over to Tool 0, even if that spool may be physically installed on a different tool. If the PrusaMMU dialog appeared first, it would allow for selecting which tool to use, then have the Spool Manager dialog appear to select which spool is physically present on that tool so that filament usage totals are correct in the end.

crysxd, 2024-03-31:
This is not easily possible due to how the two plugins work. OctoApp needs to tell SpoolManager before the print what spool is supposed to be used, this is why the dialog is shown by the app as part of the print start sequence. The dialog from the MMU plugin is not managed by the app, it's managed by the plugin itself. It sends a signal to the app, telling it to show the dialog. In many cases this will be at the start of the print, but it can also happen during the print.

What I could do is not show the SpoolManager dialog at all when the MMU plugin is installed....but this might break other things

Java-SE-Binary, 2024-03-31:
Not showing the SpoolManager dialog is how the web interface gets configured when the MMU plugin is installed, it will only prompt the user to select which tool to use, with no prompt from SpoolManager. 

If the SpoolManager dialog were to be disabled, I feel like the option to select material on a per tool basis would need to be added to the materials menu, since as it stands the materials menu only changes which material exists on Tool 0.

crysxd, 2024-03-31:
But if I remove the SpoolManager dialog, is the MMU plugin activating the correct Spool?

Java-SE-Binary, 2024-03-31:
It should be, yes. As long as the user assigns the correct spools to the correct tools before starting the print, PrusaMMU communicates which tool is being used and SpoolManager will then use the spool that was assigned to that tool by the user. Thats why having the extra ability to assign spools to each of the 5 tools would be important if this change were to be implemented.

On top of this, PrusaMMU will hide any tools that do not have spools assigned to them from SpoolManager when the dialog shows, so its impossible to select an empty tool and have usage totals go to waste.

crysxd, 2024-04-02:
Hiding the dialog if one of the two MMU plugins is installed is not a big deal. The only thing that makes me hesitate is that you are the first one to report this in over 2 years...

I will look into this a bit more later this week

Can you shoot me a mail to hello@octoapp.eu? I do not offer support via GitHub normally, I want everything in one support bucket (otherwise I go crazy at some point :D)

Java-SE-Binary, 2024-04-02:
Gotcha, no worries! I'll shoot you an email and close this out in the meantime!

closed 2024-04-02 by Java-SE-Binary