crysxd/OctoApp-Plugin #68: Issue when installing OctoApp for Klipper

joshtheclipper, 2024-04-06:
## Please do not open tickets with bugs. Use the app's bug report feature instead (main menu > FAQ & help > report a bug)
I am unable to install the plugin I am getting this error:

Err:6 http://armbian.lv.auroradev.org/apt buster Release
  404  Not Found [IP: 23.186.113.5 80]
Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

crysxd, 2024-04-06:
That's a problem with your Linux, not the plugin. The installer tries to `apt-get update` which most likely results in this error.

Googeling the error leads me to this page: [he repository 'http://apt.armbian.com/ buster Release' no longer has a Release file.](https://forum.openmediavault.org/index.php?thread/51664-http-apt-armbian-com-buster-release-does-not-have-a-release-file/)

> Armbian pulled the files for older releases. We can't do anything about that

Basically your OS is outdated

closed 2024-04-06 by crysxd

joshtheclipper, 2024-04-06:
Thank you for the prompt response. I am very new to all this. I just took out that line and the install went just fine.