crysxd/OctoApp-Plugin #71: Repository is missing a Release File ?

Cookie-fighter, 2024-04-11:
When going through KIAUH and/or directly via bash command.
The Process end with these Lines:

Reading package lists... Done
E: The repository 'http://apt.armbian.com buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

as i´m new to all of this SSH stuff (and mostly GitHub) i have no idea what more informations you need so let me know
and i will try to provide what is needed

closed 2024-04-11 by Cookie-fighter

crysxd, 2024-04-12:
Jut out of curiosity, how did you solve it?
The underlying issue is your OS being outdated.

Cookie-fighter, 2024-04-13:
Hey
Thanks for asking.
Actually I didn’t solve it but I found the „issue“ to be answered before. Wich I missed earlier. Therefor I closed it as no one needs the same question answered over and over ^^

crysxd, 2024-04-13:
Perfect! I actually pushed a mini update of the installer a few mins ago that makes the `sudo apt update` step optional, so even if the update process doesn't work anymore on bullseye the installer will continue

Cookie-fighter, 2024-04-19:
Thanks for the information I gonna give it another try