crysxd/OctoApp-Plugin #73: Octoapp Companion for K1 Max not installing

KessTheKing, 2024-05-10:
## Please do not open tickets with bugs. Use the app's bug report feature instead (main menu > FAQ & help > report a bug)
Running in K1 and K1 Max OS mode
Checking required system packages are installed...
Requirement already satisfied: virtualenv in /usr/lib/python3.8/site-packages (15.1.0)
System package install complete.
Checking Python Virtual Environment For OctoApp...
Updating PIP if needed... (this can take a few seconds or so)
Could not fetch URL https://pypi.python.org/simple/pip/: There was a problem confirming the ssl certificate: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108) - skipping
Requirement already up-to-date: pip in /usr/data/octoapp-env/lib/python3.8/site-packages
Installing or updating required python libs...
  Could not find a version that satisfies the requirement websocket_client<1.6.99,>=1.6.0 (from -r /usr/data/octoapp/requirements.txt (line 10)) (from versions: )
No matching distribution found for websocket_client<1.6.99,>=1.6.0 (from -r /usr/data/octoapp/requirements.txt (line 10))

grubfriend, 2024-05-11:
Same problem here. I manually installed websocket_client via command line (python3 -m pip install websocket-client, which installed websocket-client-1.8.0) to see if that would help, but running the octoapp install script gave me the same error.

crysxd, 2024-05-12:
Sorry, I wrote a comment yesterday and didn't send it...

I updated the installer to a different version of the Websocket lib, I see @KessTheKing you tried again with the new version and it still happens.

Looking at the logs again, I think the root cause is `certificate verify failed: certificate has expired`, which prevents pip from loading the library.

@grubfriend You used the global python, but OctoApp has it's onw at `/usr/data/octoapp-env/bin/python3`

Can you verify that the `date` command shows the current time and date?

crysxd, 2024-05-12:
Not sure if an incorrect `date` is the cause, if I change mine purposefully to the past, it says `not valid yet` instead of expired and if I set 2026 the logs look different:

```
Running in K1 and K1 Max OS mode
Checking required system packages are installed...
Requirement already satisfied: virtualenv in /usr/lib/python3.8/site-packages (15.1.0)
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))) - skipping
System package install complete.
Checking Python Virtual Environment For OctoApp...
Updating PIP if needed... (this can take a few seconds or so)
Requirement already satisfied: pip in /usr/data/octoapp-env/lib/python3.8/site-packages (24.0)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))': /simple/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))': /simple/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))': /simple/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))': /simple/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))': /simple/pip/
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))) - skipping
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108)'))) - skipping
```

grubfriend, 2024-05-12:
> Can you verify that the `date` command shows the current time and date? The certificate in question is valid from 2024-04-23 till 2025-05-25
Yep, Date returned Sun May 12 17:17:22 AEST 2024. Could it be something on the virtual environment end?

crysxd, 2024-05-12:
I honestly don't know, I wouldn't know what in a venv could influence the cert to be expired. I'll look into this a bit more later today

crysxd, 2024-05-12:
@grubfriend I'm really lost here, the fact that I can't reproduce this issue is really a show stopper for me at the moment. Can you try following things? I just hope that we find something to point us in the right direction

- `/usr/data/octoapp-env/bin/python3 -m pip install -r /usr/data/octoapp/requirements.txt` installs all packages as the installer
- `python3 -m pip install -r /usr/data/octoapp/requirements.txt` does the same but with the system python. Make sure to clean up with `/usr/data/octoapp-env/bin/python3 -m pip uninstall -r /usr/data/octoapp/requirements.txt`

grubfriend, 2024-05-12:
> * `python3 -m pip install -r /usr/data/octoapp/requirements.txt` does the same but with the system python. Make sure to clean up with `/usr/data/octoapp-env/bin/python3 -m pip uninstall -r /usr/data/octoapp/requirements.txt`

@crysxd this second one worked! Install completed and the app is working. I got an error for pycryptodome during install but otherwise it's fine. Here's the pycryptodome error just in case:

```Attempting to install optinal python libs...
  error: subprocess-exited-with-error
  × Building wheel for pycryptodome (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [343 lines of output]
      Testing support for clang
      Target does not support clang
      Testing support for gcc
      Target does not support gcc
      Testing support for stdint.h header
      Target does not support stdint.h header
      Testing support for 128-bit integer
      Target does not support 128-bit integer
      Testing support for cpuid.h header
      Target does not support cpuid.h header
      Testing support for intrin.h header
      Target does not support intrin.h header
      Testing support for posix_memalign
      Target does not support posix_memalign
      Testing support for memalign
      Target does not support memalign
      Testing support for SSE2(intrin.h)
      Target does not support SSE2(intrin.h)
      Testing support for SSE2(x86intrin.h)
      Target does not support SSE2(x86intrin.h)
      Testing support for SSE2(emmintrin.h)
      Target does not support SSE2(emmintrin.h)
      Warning: compiler does not support AESNI instructions
      Warning: compiler does not support CLMUL instructions
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-mips-3.8
      creating build/lib.linux-mips-3.8/Crypto
      copying lib/Crypto/__init__.py -> build/lib.linux-mips-3.8/Crypto
      creating build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_pkcs1_oaep_decode.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_siv.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_openpgp.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_ofb.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_ocb.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_gcm.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_ecb.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_eax.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_ctr.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_cfb.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_ccm.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_mode_cbc.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/__init__.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/_EKSBlowfish.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/Salsa20.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/PKCS1_v1_5.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/PKCS1_OAEP.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/DES3.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/DES.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/ChaCha20_Poly1305.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/ChaCha20.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/CAST.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/Blowfish.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/ARC4.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/ARC2.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      copying lib/Crypto/Cipher/AES.py -> build/lib.linux-mips-3.8/Crypto/Cipher
      creating build/lib.linux-mips-3.8/Crypto/Hash
      copying lib/Crypto/Hash/keccak.py -> build/lib.linux-mips-3.8/Crypto/Hash
      copying lib/Crypto/Hash/cSHAKE256.py -> build/lib.linux-mips-3.8/Crypto/Hash
      copying lib/Crypto/Hash/cSHAKE128.py -> build/lib.linux-mips-3.8/Crypto/Hash
[... 263 lines cut ...]
      copying lib/Crypto/Util/asn1.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Util/_raw_api.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Util/_file_system.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Util/_cpu_features.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Util/RFC1751.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Util/Padding.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Util/Counter.pyi -> build/lib.linux-mips-3.8/Crypto/Util
      copying lib/Crypto/Math/_IntegerNative.pyi -> build/lib.linux-mips-3.8/Crypto/Math
      copying lib/Crypto/Math/_IntegerGMP.pyi -> build/lib.linux-mips-3.8/Crypto/Math
      copying lib/Crypto/Math/_IntegerCustom.pyi -> build/lib.linux-mips-3.8/Crypto/Math
      copying lib/Crypto/Math/_IntegerBase.pyi -> build/lib.linux-mips-3.8/Crypto/Math
[... cut ...]
      error: command '/opt/huangyushan/ingenic_linux_master/buildroot/buildroot/output/host/bin/mips-linux-gnu-gcc' failed with exit status 1
      [end of output]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycryptodome
ERROR: Could not build wheels for pycryptodome, which is required to install pyproject.toml-based projects
Failed to install optional python libraries. Optional end-to-end encryption for Android notification is not possible. Continuing...
```

crysxd, 2024-05-12:
pycryptdome doesn't work on K1.

I'm a bit confused why installing with the system python3 influenced the installer...

KessTheKing, 2024-05-12:
@crysxd I believe it worked.  The steps I did.

I followed the manual install, I did receive the original error. 
cd /usr/data
git clone https://github.com/crysxd/OctoApp-Plugin octoapp
cd octoapp
./install.sh 

I then did this - python3 -m pip install -r /usr/data/octoapp/requirements.txt (received no errors) once this was complete I did the manual install - I had to use "sh ./install.sh"

Everything installed and seems to be working.  

Running in K1 and K1 Max OS mode
Checking required system packages are installed...
Requirement already satisfied: virtualenv in /usr/lib/python3.8/site-packages (15.1.0)
System package install complete.
Checking Python Virtual Environment For OctoApp...
Updating PIP if needed... (this can take a few seconds or so)
Requirement already satisfied: pip in /usr/data/octoapp-env/lib/python3.8/site-packages (24.0)
Installing or updating required python libs...
Python libs installed.
Bootstrap done. Starting python installer...
Os Type Detected: OsTypes.K1

Only one moonraker instance was found, so we are using it! [S56moonraker_service:/usr/data/printer_data/config/moonraker.conf]
Starting configuration...
Enuring path and permissions [/usr/data/printer_data/octoapp-store]...
Dir already exists.
Setting owner permissions to the service user [root]...
Directory setup successfully.
Configured. Service: S66octoapp_service, Path: /etc/init.d/S66octoapp_service, LocalStorage: /usr/data/printer_data/octoapp-store, Config Dir: /usr/data/printer_data/config, Logs: /usr/data/printer_data/logs
Starting Web Interface Setup

The following web interfaces were automatically discovered:
  1) Fluidd   - Port 4408
  2) Fluidd   - Port 80

Enter the number next to the web interface you would like to use, or enter `m` to manually setup the web interface: 1
Setting Up OctoApp's System Service...
Creating service run script...
Making the run script executable...
Creating service file /etc/init.d/S66octoapp_service...
Making the service executable...
Starting the service...
Service setup and start complete!

closed 2024-05-13 by crysxd