crysxd/OctoApp-Plugin #80: Opening and closing Android app automatically turns on or off printer

cheribulm, 2024-06-26:
Hello,
If I go to monitor my printer and then close the app on my table it turns off my printer. Likewise, if the printer is off and I open the app the printer turns on.
I do have the WeMo plugin installed on my Octopi. Even if I turn off all the functions of the plugin it will still do this.
I have searched all over the app to find a way to disable it to no avail. If I can't find a solution I'll be forced to abandon the app because I've ruined so many print jobs going into the app, closing it, and having the printer turn off.

Thanks for your assistance.

crysxd, 2024-06-27:
As mentioned in the template, please don't open tickets with bugs but use the app's bug report feature in main menu > Settings FAQ & Help > Report a bug.

In your case, I bet you enabled the automatic lights feature and pointed it to the printer instead of a light. Check main menu > Settings > Power Devices > Automatic lights. If that's not the case, please send a bug report as described above.

closed 2024-06-27 by crysxd

cheribulm, 2024-06-28:
Thank you for your fast response despite my ability to follow instructions
properly. As well your solution was right on the money and has saved me
from tons of frustration.

Thanks again,
Gabe-

crysxd, 2024-06-29:
Glad it's fixed! Could you tell me a bit more how this happened? What could the app have done to communicate this better to you?

Usually I really need a bug report to figure stuff out ;) it contains not only the app logs but also settings.

cheribulm, 2024-06-29:
Hello,
I would say it is the approach.
If you say hey would you like your LED lights to turn on automatically when
you open the app to see what is happening, then turn off when you close the
app? My answer would be yes that sounds amazing.
But the same being true, Hey would you like your PSU/ printer to turn on
and off when you use the app? I would tell you to go to hell you sick
bastard.
But these functions are the same if the end-user (me) does not know the
nuances of it. Having the ability to do both is possible if one is not
aware of it.
I enabled the first function not realizing I was doing the second as I
believe in one of the tooltips it offered to enable this I did not actually
go to that location main menu > Settings > Power Devices > Automatic lights
Or I would have noticed the large note there detailing exactly what
happened.
So I would approach this in two ways Take the tooltip away to have this
function done automatically. And two, be redundant with the settings and
keep the Automatic lights one the way it is as it clearly does what it
says. And then make a separate function to automatically turn off the PSU/
printer in idle or whatever. If you want to be super cautious you could
have a popup notice when selecting the device for the Automatic lights
warning them to be careful and do not select your printer but the device
the lights are plugged into.

I hope this explanation helps. If you need more clarity let me know. i am
writing this at 5am so my mind might be a little foggy ;)

Gabe-

cheribulm, 2024-06-29:
Hi, Since I had you ear for the moment. I was wondering if I could make a
request for a function?
I have a Arducam B0283 Camera Pan Tilt Platform that has no real support.
There is this demo app that works on Git but none of the plugins on
OctoPrint work on controlling it.
Is there a way you could look at the code of the demo and insert the
controls into your app so people (me) can finally use this to control their
cameras in Octoipi. It would be another great incentive to for folks to buy
your app as you would be the only one with this function (wink).

ArduCAM demo <https://github.com/ArduCAM/PCA9685/tree/master/example/rpi>

Arducam Upgraded Camera Pan Tilt Platform
<https://www.amazon.com/dp/B08PK9N9T4>

Thanks again
Gabe-

crysxd, 2024-06-30:
The only way to enable the automatic lights is via the menu with the big warning, just double checked the code...there is no other way. There is no automatic enabling of this - for good reason :D 

So the only way I see is to make the warning even bigger....

The app can't know what a certain power device controls. Some are know to only be lights (e.g. WS2812 plugin), but many can be a light, an external fan or the printer itself. This means I also can't filter certain devices out.

crysxd, 2024-06-30:
Regarding the Arduino cam, I made a note. So far you are the only person to ask for this, so I will see if there is more demand

cheribulm, 2024-06-30:
No problem, I just wanted to share that as it is the only out-of-the-box
camera PTZ controllers for the Pi and it would work well for your app.

Thanks again for the help on the other issue
Gabe-