crysxd/OctoApp-Plugin #99: OctoApp not connecting from android, but works on iPhone

NatroNx, 2025-01-03:
Hello Christian,

this is probably not the right channel as it isn't directly connected to the plugin but I found no better way to contact you.

I'm using OctoApp for years now on both android and Ios and got a new printer now and found an Issue. On my Anycubic Kobra 3 with DuckPro Firmware (Klipper) , the connection from my Iphone works just fine while my Android phone hangs at the "Testing the connection to your printer..." screen.

Any Ideas or troubleshooting I can do?

crysxd, 2025-01-03:
This is not the right channel :D 
Please send a bug report from main menu > Settings > FAQ & help  > Report a bug. On the testing screen you can keep the "Need help" pressed to do the same.

My guess is that you use .local but Android is really bad with it. Try connecting using a static IP address instead, this most likely solves the problem. If not, shoot me the bug report and we check it out there.

crysxd, 2025-01-03:
As a side note, if I don't respond to the bug report over the weekend ping me here again...my email inbox is making trouble right now

closed 2025-01-03 by crysxd