crysxd/OctoApp-Plugin #101: [Creality K1/Klipper] Error while installing OctoApp plugin in sh

cdunn95, 2025-01-15:
When attempting to install the OctoApp through the [Helper Script](https://guilouz.github.io/Creality-Helper-Script-Wiki/helper-script/octoapp-companion/), I kept running into this issue:

```
Running in K1 and K1 Max OS mode
Checking required system packages are installed...
Downloading http://bin.entware.net/mipselsf-k3.4/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Package python3 (3.11.7-1) installed in root is up to date.
Package python3-pillow (10.1.0-1) installed in root is up to date.
Package python3-pip (23.3.1-1) installed in root is up to date.
Requirement already satisfied: virtualenv in /usr/lib/python3.8/site-packages (15.1.0)
System package install complete.
Checking Python Virtual Environment For OctoApp...
No virtual environment found, creating one now.
./install.sh: line 191: /opt/bin/virtualenv: not found
```

After poking around with my (novice) linux terminal abilities, I eventually found the script itself in `/usr/data/octoapp` and attempted to install it standalone by executing `sh ./install.sh`, which gives me the same error as above.

To ensure that I am on the correct release branch, I checked the `git log` and can confirm I am on the latest `origin/release`, with commit hash 612d03c73ae0f577a9b0eae8c097fcd8e8a37a06 .

As stated previously, I am fairly noob with linux terminals, and also not super acquainted with Python. If there is something simple I am overlooking I apologize. Please let me know if there is anything that I can do to help troubleshoot and/or resolve this issue. 

Thanks in advance, and thanks for making such a gorgeous application!

crysxd, 2025-01-15:
That seems to be caused by a fix I did for another K1 user. I officially hat this machine, it's an absolute game of whac-a-mole. 

I made a fix for the fix, please run `rm -rf /usr/data/octoapp` and try again from the start

closed 2025-01-15 by crysxd

crysxd, 2025-01-16:
@cdunn95 Did you have a chance to check if it works now? :) Would be great to get a confirmation!

crysxd, 2025-01-18:
Amazing :)

Regarding the OE stuff, that's a good hint. I should remove the explicit numbers here because I don't actively track changes. The idea was to give the users a quick overview

CleverCompiler, 2025-03-12:
I was having the same issue after upgrading to the 1.3.46 firmware.  I did a factory reset before upgrading and installed the K1 helper script from scratch.  I found the answer after running the reset in the K1 helper script which did a more though reset.

CleverCompiler, 2025-03-13:
Yes. Resetting with the helper script cleared whatever was causing my error.