crysxd/OctoApp-Plugin #113: Sovol Zero octoapp issue

cpuks, 2025-04-11:
I'm trying to install Octoapp via KIAUH and that's an error I get:
```
sovol@SPI-XI:~$ /home/sovol/octoapp-env/bin/pip install requests certifi
Requirement already satisfied: requests in ./.local/lib/python3.9/site-packages (2.32.3)
Requirement already satisfied: certifi in ./.local/lib/python3.9/site-packages (2024.8.30)
Requirement already satisfied: charset-normalizer<4,>=2 in ./.local/lib/python3.9/site-packages (from requests) (3.4.0)
Requirement already satisfied: idna<4,>=2.5 in ./.local/lib/python3.9/site-packages (from requests) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./octoapp-env/lib/python3.9/site-packages (from requests) (1.26.20)
sovol@SPI-XI:~$ /home/sovol/octoapp/install.sh

                                          .....
                                   .'^":IIIIIIIII;,"`'.
                               .'";IIII;,,""^^"",:IIIII:^.
                             '"III;"`'.............'`^,III;`.
                           ',III"`......................`:III^
                          ^III"'..........................`:II;'
                        .,II:'..............................^III`
                        :II,.................................`III`
                       "II:...................................`III.
                      .III`....................................,II,
                      `II;...........'`'..........'`'..........`III.
                      ^II;.........':IIII`......`;III:'........`III.
                      `III.........`IIIII,......"IIIII`........^II;
                      .III`.........`",,^........`,,"`.........;II^
                       ^II:...................................`III.
                       .III'..................................,II,
                        ;II`..................................;II`
                        :II^..................................III`
                        III`..................................:II^
                       `II;...................................^II;.
         `"^.         .;II^....................................;II^          `"".
         "III^.       ,II:....''...........................`'..`III`       ':II;.
          ':III"'.  ':II:'..`;II'.........................:II,..`III".  .`:III^
            ',IIIIIIIII"...^III^...^^'.....'```'....',,'..';II,'.',IIIIIIII:^.
             ..`^,,,"`...':II;`..';II"..`,IIIIII:`..^II;'...,II;`..'`",""`'...
             ..........'"III"...`;II".."III^''^;II:'.,II;'...`;II:`...........
             .'.....'`,III,'...^III^..,II;.    ."II;`."II;`...."III;^`.....''
            :II;:,:IIII;^'...',II;`.."II;.       ^III'.^III,'....`:IIIII;;III`
            .`":;I;:"^'.....^III,'.."II;.         "II;'.',III"'....'`^,,:,,^'
                 ........`"III;`..',II;.           "II;`..`:III,^'........
                `"""",,;IIII,`..'"III".             `III,'..'^:IIII;::::;;^
                "IIIII;:,^`...`,III,.                ."III,`...'`^",::::,"'
                    ........';II;^.                    ."IIII'...
                             '`'                          '``

                                   OctoApp For Klipper
                            Based on the OctoEverywhere Plugin

Checking required system packages are installed...
You might be asked for your system password - this is required to install the required system packages.
Fri 11 Apr 2025 01:33:37 PM CEST
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:3 http://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Get:4 http://deb.debian.org/debian-security bullseye-security InRelease [27.2 kB]
Get:5 https://pkgs.tailscale.com/stable/debian bullseye InRelease
Fetched 127 kB in 2s (52.7 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (20.3.4-4+deb11u1).
virtualenv is already the newest version (20.4.0+ds-2+deb11u1).
python3 is already the newest version (3.9.2-3).
curl is already the newest version (7.74.0-1.3+deb11u14).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Ensuring zlib is install for Pillow, it's ok if this package install fails.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2+deb11u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
System package install complete.
Checking Python Virtual Environment For OctoApp...
Updating PIP if needed... (this can take a few seconds or so)
Requirement already satisfied: pip in ./octoapp-env/lib/python3.9/site-packages (25.0.1)
Installing or updating required python libs...
Python libs installed.
Attempting to install optional python libs...
Python libs installed.
Bootstrap done. Starting python installer...
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/sovol/octoapp/moonraker_installer/__main__.py", line 2, in <module>
    from .Installer import Installer
  File "/home/sovol/octoapp/moonraker_installer/Installer.py", line 10, in <module>
    from .Configure import Configure
  File "/home/sovol/octoapp/moonraker_installer/Configure.py", line 5, in <module>
    from octoapp.websocketimpl import Client
  File "/home/sovol/octoapp/octoapp/websocketimpl.py", line 2, in <module>
    import certifi
ModuleNotFoundError: No module named 'certifi'
```

cpuks, 2025-04-12:
Os some progress:
```
sovol@SPI-XI:~$ /home/sovol/octoapp-env/bin/pip install --no-cache-dir --force-reinstall certifi
Collecting certifi
  Downloading certifi-2025.1.31-py3-none-any.whl.metadata (2.5 kB)
Downloading certifi-2025.1.31-py3-none-any.whl (166 kB)
Installing collected packages: certifi
  Attempting uninstall: certifi
    Found existing installation: certifi 2024.8.30
    Not uninstalling certifi at /home/sovol/.local/lib/python3.9/site-packages, outside environment /home/sovol/octoapp-env
    Can't uninstall 'certifi'. No files were found to uninstall.
```
And now it throws error:
```
Bootstrap done. Starting python installer...
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/sovol/octoapp/moonraker_installer/__main__.py", line 2, in <module>
    from .Installer import Installer
  File "/home/sovol/octoapp/moonraker_installer/Installer.py", line 12, in <module>
    from .Permissions import Permissions
  File "/home/sovol/octoapp/moonraker_installer/Permissions.py", line 6, in <module>
    from .Frontend import Frontend
  File "/home/sovol/octoapp/moonraker_installer/Frontend.py", line 4, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```
So looks like it has problem to properly setup venv to build plugin

crysxd, 2025-04-12:
So it's literally this:

```
/home/sovol/octoapp-env/bin/pip install --no-cache-dir --force-reinstall requests certifi
/home/sovol/kiauh/kiauh
```

Interesting. I remember there was something with the SV08 that made the OS a bit "funny" and some people flashed the CB1 OS (same chip). Not sure if this was this issue.

So everything is up and running now?

Regarding changes, my problem is that I can't verify that this is your Zero or all Zeros and I'm also not sure how to detect this machine...

cpuks, 2025-04-12:
Yeah exactly that does the trick.
I just dropped link to github onto sovol zero discord channel so probably
more people will report back (45) Discord | #zero-series-printers | Sovol
3D Printers
<https://discord.com/channels/788873913953157131/1351950438043222179>

crysxd, 2025-04-14:
I made some changes specific to Sovol. Can you please test them?

```
/home/sovol/octoapp-env/bin/pip uninstall -y requests certifi        # Undo fix
cd                                                                                                          # Go home
rm -rf octoapp octoapp-env                                                              # Kill OctoApp
git clone https://github.com/crysxd/OctoApp-Plugin octoapp.       # Clone OctoApp
cd octoapp                                                                                           # Open OctoApp dir
git checkout sovol                                                                               # Switch to branch with fix
./install.sh                                                                                             # Run installer
```

In the output you should see`Running in Sovol mode` and `Additional python libs for Sovol installed.`

crysxd, 2025-04-18:
Will merge it. Thanks!

closed 2025-04-18 by crysxd