crysxd/OctoApp-Plugin #117: Plugin installation issue

ProfessionalBawkBawkGames, 2025-05-06:
Running in K1 and K1 Max OS mode
Checking required system packages are installed...
Requirement already satisfied: virtualenv in /usr/lib/python3.11/site-packages (20.31.1)
Requirement already satisfied: distlib<1,>=0.3.7 in /usr/lib/python3.11/site-packages (from virtualenv) (0.3.9)
Requirement already satisfied: filelock<4,>=3.12.2 in /usr/lib/python3.11/site-packages (from virtualenv) (3.18.0)
Requirement already satisfied: platformdirs<5,>=3.9.1 in /usr/lib/python3.11/site-packages (from virtualenv) (4.3.7)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
System package install complete.
Checking Python Virtual Environment For OctoApp...
No virtual environment found, creating one now.
python3: can't open file '/usr/lib/python3.8/site-packages/virtualenv.py': [Errno 2] No such file or directory
AD5M ~/octoapp #

crysxd, 2025-05-07:
Please run `ls /usr/lib/`

K1 is horrible :( The OS is just half broken all the time

ProfessionalBawkBawkGames, 2025-05-07:
which venv doesnt seem to do anything

crysxd, 2025-05-07:
And `ls /usr/lib/`?
If `which venv` doesn't do anything `venv` is not installed

Can you give me a bit more context how you are trying to install the plugin? At the moment you just posted some logs and told me venv doesn't do anything, that's not a lot of info for me to help

ProfessionalBawkBawkGames, 2025-05-07:
ls /usr/lib/ looks like tosses out a massive directly like i guess you would comparatively see with dir in command prompt. so basically I copied the repo went into the directory

AD5M ~ # ls
octoapp           printer_data      printer_software
AD5M ~ #

and once inside the directory i just run sh install.sh

ProfessionalBawkBawkGames, 2025-05-07:
AD5M ~ # ls /usr/lib/
bfd-plugins                         libgmp.so.10.4.1                    libpcap.so.1
device-mapper                       libgmpxx.so                         libpcap.so.1.10.3
libaio.so                           libgmpxx.so.4                       libpng.so
libaio.so.1                         libgmpxx.so.4.6.1                   libpng16.so
libaio.so.1.0.2                     libgnutls.so                        libpng16.so.16
libavcodec.so                       libgnutls.so.30                     libpng16.so.16.39.0
libavcodec.so.58                    libgnutls.so.30.37.1                libpopt.so
libavcodec.so.58.134.100            libgnutlsxx.so                      libpopt.so.0
libavdevice.so                      libgnutlsxx.so.30                   libpopt.so.0.0.2
libavdevice.so.58                   libgnutlsxx.so.30.0.0               libpython3.11.so
libavdevice.so.58.13.100            libgpiod.so                         libpython3.11.so.1.0
libavfilter.so                      libgpiod.so.2                       libpython3.so
libavfilter.so.7                    libgpiod.so.2.2.2                   libreadline.so
libavfilter.so.7.110.100            libgpiodcxx.so                      libreadline.so.8
libavformat.so                      libgpiodcxx.so.1                    libreadline.so.8.2
libavformat.so.58                   libgpiodcxx.so.1.1.2                libsmartcols.so
libavformat.so.58.76.100            libhistory.so                       libsodium.so
libavutil.so                        libhistory.so.8                     libsodium.so.23
libavutil.so.56                     libhistory.so.8.2                   libsodium.so.23.3.0
libavutil.so.56.70.100              libhogweed.so                       libsqlite3.so
libbfd-2.38.so                      libhogweed.so.6                     libsqlite3.so.0
libbfd.so                           libhogweed.so.6.6                   libsqlite3.so.0.8.6
libblkid.so                         libip4tc.so                         libssl.so
libbsd.so                           libip4tc.so.2                       libssl.so.1.1
libbsd.so.0                         libip4tc.so.2.0.0                   libstdc++.so.6
libbsd.so.0.11.3                    libip6tc.so                         libstdc++.so.6.0.29
libcamera-base.so                   libip6tc.so.2                       libstdc++.so.6.0.29-gdb.py
libcamera-base.so.0.0.4             libip6tc.so.2.0.0                   libswresample.so
libcamera.so                        libjpeg.so                          libswresample.so.3
libcamera.so.0.0.4                  libjpeg.so.8                        libswresample.so.3.9.100
libcrypto.so                        libjpeg.so.8.2.2                    libswscale.so
libcrypto.so.1.1                    libkeyutils.so                      libswscale.so.5
libctf-nobfd.so                     libkeyutils.so.1                    libswscale.so.5.9.100
libctf-nobfd.so.0                   libkeyutils.so.1.10                 libtasn1.so
libctf-nobfd.so.0.0.0               liblvm2cmd.so                       libtasn1.so.6
libctf.so                           liblvm2cmd.so.2.03                  libtasn1.so.6.6.3
libctf.so.0                         liblzma.so                          libtirpc.so
libctf.so.0.0.0                     liblzma.so.5                        libtirpc.so.3
libcurl.so                          liblzma.so.5.2.10                   libtirpc.so.3.0.0
libcurl.so.4                        libmagic.so                         libts.so
libcurl.so.4.8.0                    libmagic.so.1                       libts.so.0
libcurses.so                        libmagic.so.1.0.0                   libts.so.0.10.4
libdbus-1.so                        libmd.so                            libturbojpeg.so
libdbus-1.so.3                      libmd.so.0                          libturbojpeg.so.0
libdbus-1.so.3.19.17                libmd.so.0.0.5                      libturbojpeg.so.0.2.0
libdevmapper-event-lvm2.so          libmenu.so                          libunistring.so
libdevmapper-event-lvm2.so.2.03     libmenu.so.6                        libunistring.so.5
libdevmapper-event-lvm2mirror.so    libmenu.so.6.4                      libunistring.so.5.0.0
libdevmapper-event-lvm2raid.so      libmount.so                         libuuid.so
libdevmapper-event-lvm2snapshot.so  libmpdec++.so                       libuv.so
libdevmapper-event-lvm2thin.so      libmpdec++.so.2.5.1                 libuv.so.1
libdevmapper-event-lvm2vdo.so       libmpdec++.so.3                     libuv.so.1.0.0
libdevmapper-event.so               libmpdec.so                         libv4l
libdevmapper-event.so.1.02          libmpdec.so.2.5.1                   libv4l1.so
libdevmapper.so                     libmpdec.so.3                       libv4l1.so.0
libdevmapper.so.1.02                libncurses.so                       libv4l1.so.0.0.0
libell.so                           libncurses.so.6                     libv4l2.so
libell.so.0                         libncurses.so.6.4                   libv4l2.so.0
libell.so.0.0.2                     libnettle.so                        libv4l2.so.0.0.0
libevdev.so                         libnettle.so.8                      libv4l2rds.so
libevdev.so.2                       libnettle.so.8.6                    libv4l2rds.so.0
libevdev.so.2.3.0                   libnfsidmap                         libv4l2rds.so.0.0.0
[... cut ...]
libffi.so.8                         libnl-xfrm-3.so.200.26.0            terminfo
libffi.so.8.1.2                     libopcodes-2.38.so                  ts
libform.so                          libopcodes.so                       udev
libform.so.6                        libpanel.so                         v4l1compat.so
libform.so.6.4                      libpanel.so.6                       v4l2-compat.so
libgmp.so                           libpanel.so.6.4                     v4l2convert.so
libgmp.so.10                        libpcap.so                          xtables

crysxd, 2025-05-07:
Did you try to install it via the K1 helper script as described here?
https://github.com/crysxd/OctoApp-Plugin/wiki/Installation-on-Creality

ProfessionalBawkBawkGames, 2025-05-07:
no i didnt see that one giving a test now and thank you

ProfessionalBawkBawkGames, 2025-05-07:
so whats weird is looking back im realizing...if k1 is in reference to the creality k1....I do not have this....i am running a Flashforge ad5m

crysxd, 2025-05-07:
That explains a lot :D That's why a lot of info is important ;)
I don't know this printer and don't know the operating system on it. If the plugin detects K1 mode then I assume it's similar.

Can you try this?
https://github.com/crysxd/OctoApp-Plugin/issues/117#issuecomment-2858510499

ProfessionalBawkBawkGames, 2025-05-07:
1c. Manual installation
Connect to your machine via SSH
Run git clone https://github.com/crysxd/OctoApp-Plugin octoapp
Run cd octoapp
Run ./install.sh
The installation is now started, follow the prompts on screen

this is what i ran to originally install it and instead of run ./install.sh which never works
i ran sh install.sh

ProfessionalBawkBawkGames, 2025-05-07:
> That explains a lot :D That's why a lot of info is important ;) I don't know this printer and don't know the operating system on it. If the plugin detects K1 mode then I assume it's similar.
> 
> Can you try this? [#117 (comment)](https://github.com/crysxd/OctoApp-Plugin/issues/117#issuecomment-2858510499)

yeah im really sorry bout that not been 100% with it lately, I am usually really good at providing tons of info with detail so this is on me doing too amny things at once

crysxd, 2025-05-07:
> 1c. Manual installation Connect to your machine via SSH Run git clone https://github.com/crysxd/OctoApp-Plugin octoapp Run cd octoapp Run ./install.sh The installation is now started, follow the [...]

Please note the `git checkout k1-install-fix`, I pushed changes for you :)
Please run:

```
rm -rf octoapp 
rm -rf octoapp-env
git clone https://github.com/crysxd/OctoApp-Plugin octoapp
cd octoapp
git checkout k1-install-fix
./install.sh
```

> yeah im really sorry bout that not been 100% with it lately, I am usually really good at providing tons of info with detail so this is on me doing too amny things at once
No worries :)

crysxd, 2025-05-07:
But before you continue. Does this machine run Klipper?
I don't see any reference to that. If it doesn't run Klipper and Moonraker you can't use the app

ProfessionalBawkBawkGames, 2025-05-07:
> But before you continue. Does this machine run Klipper? I don't see any reference to that. If it doesn't run Klipper and Moonraker you can't use the app

yes im running klippermod right now

crysxd, 2025-05-07:
Step by step, at least venv is working now

You can use `df -h` to see which file system has free space. OctoApp tries to install at `/usr/data`
Make sure to run `rm -rf octoapp` and `rm -rf octoapp-env` to clean up

ProfessionalBawkBawkGames, 2025-05-07:
oh yeah tried these others...cant seem to get htem to install either or get rid of them
AD5M ~ # ls /usr/data
mobileraker-env        mobileraker_companion  octoapp-env            octoeverywhere         octoeverywhere-env

ProfessionalBawkBawkGames, 2025-05-07:
AD5M ~ # ls /mnt/data
100+x+17.5+mm,+Push-Fit+Hook_PLA_23m31s.gcode                      Keychain_PLA_4m7s.gcode                                            bradnailer.gcode
100mm_VESA_Mount.gcode                                             Knob Wrench.gcode                                                  camera
20 mm Big Thread - Plain Head - Bolt.gcode                         Mobile phone holder_PLA_39m30s.gcode                               cardinaloutlet v1_PLA_1h58m.gcode
20mm_Box.gx                                                        Moderate Weight Bearing Snap.gcode                                 dragonegg-sturdier-bottom_PLA_2h6m.gcode
3DBenchy.gcode                                                     Moderate Weight Bearing Snap_PLA_12m37s.gcode                      drawers and latches.gcode
3DBenchy_ABS_41m7s.gcode                                           Mount Camera Screw.gcode                                           flowrate_0_PLA_33m36s.gcode
3dbenchy_PLA_36m33s.gcode                                          Mount Camera v5.gcode                                              handle_PLA_40m54s.gcode
3mf                                                                Mount Holder Horizontal.gcode                                      holder test.gcode
8 mm - Quad Offset Snaps DS Part A (One Piece)_PLA_50m4s.gcode     Mount Phone Clamp B with Tab v6.gcode                              holdertest_PLA_1h51m.gcode
8 mm - Quad Offset Snaps DS Part A (One Piece)_PLA_57m3s.gcode     Mount Phone Clamp Bolt v1.gcode                                    lidsLocks.gcode
8x8 Multiboard Core Tile.gx                                        Multiboard Gridfinity Drawer Chest-5x3 v2 - No Base.gcode          loading-tool_PLA_1h33m.gcode
8x8 Multiboard Core Tile_PLA_3h53m.gcode                           Offset Snaps Removal Tool_ASA_17m49s.gcode                         logFiles
8x8 Multiboard Side Tile.gcode                                     OrcaToleranceTest_ABS_10m3s.gcode                                  logcartest1-log_car_PLA_3h21m.gcode
8x8 Multiboard Side Tile_PLA_3h54m.gcode                           PYC3D-MUSHROOM-MARIO-BROS-KEYCHAIN_PLA_7m4s.gcode                  manatee test v1_PLA_2h29m.gcode
9 mm Big Thread, Flat Head, Shank Bolt.gcode                       Pegboard 8mm post v1.gcode                                         multiboard screwdriver holder shelf v2_PLA_2h58m.gcode
AD4_Airduct_Cover .gx                                              Print In Place Phone Holder.gcode                                  nozzlebrush_ad5m (1).gcode
AD5M_Camera_Mount.gcode                                            Quad Snap Connector (DS Part B) - Tight_PLA_22m13s.gcode           sawsall holder multiboard v1.gcode
Arm 150 Straight v5.gcode                                          Quad Snap Connector (DS Part B) - Tight_PLA_44m11s.gcode           sawsall holder test.gcode
Arm Axis Changer v5.gcode                                          RODSLOTH.gcode                                                     scrapertest_PLA_39m33s.gcode
Articulated Dragon.gcode                                           Rack.gcode                                                         side2.gcode
Baby_Yoda_v2.2_ABS_1h40m.gcode                                     Ryobi Tool Mount Multi Board.gcode                                 simple_calipers_with_depth_gauge_PLA_1h0m.gcode
Base Flat v5.gcode                                                 Ryobi orbital sander mount for Multiboard.gcode                    something something test.gcode
Big, Coin Slot.gcode                                               Ryobi+Charger+Holder+Multiboard.gcode                              sunset beach rocker decora v1_PLA_3h3m.gcode
Bit+Holder+Mount.gcode                                             Ryobi-mount_PLA_1h47m.gcode                                        temperature_tower.gcode
Bit+Holder+Mount_PLA_2h37m.gcode                                   RyobiToolHanger.gcode                                              temperature_tower_PLA_1h25m.gcode
Bit+Storage+With+Sign_PLA_1h47m.gcode                              Small-spool clip_PLA_11m50s.gcode                                  test.gcode
Boat_PLA_14m3s.gcode                                               SmartTemperatureTower_PLA_Plus_195-235_PLA_1h7m.gcode              testing.gcode
Chicken-my3dprintforge-v2.g                                        Smash Dice Tower V2.2_PLA_26m31s.gcode                             tolerance-test-small_PLA_13m19s.gcode
Concave Dodecahedron_PLA_38m12s.gcode                              Touch Pen_PLA_41m14s.gcode                                         towerspring.gcode
Cover_ABS_36m5s.gcode                                              Turtle Sandy.gcode                                                 trash bin_PLA_1h52m.gcode
Croc Jibbitz test v1_PLA_10m29s.gcode                              UniversalGreeblieBadge1138_PLA_38m24s.gcode                        ttr card holder v1_PLA_2h42m.gcode
Cube-PLA-Test.gcode                                                all-parts.gx                                                       ttr card holder v1_PLA_2h4m.gcode
Desk Oragnizer 60percent_PLA_57m28s.gcode                          anti-vibration-fa5m-v2-small.gcode                                 turtle outlet v1.1_PLA_2h4m.gcode
[... cut ...]
Icecream_PLA_1h2m.gcode                                            bin clips and tops.gcode                                           vertical_filament_spool_parts_drawer-dbbe-v3.gcode
ImageToStl.com_Ryobi+battery+holder+for+4mm+Clips_PLA_49m2s.gcode  blade_PLA_1h21m.gcode                                              vertical_filament_spool_parts_drawer.gcode
JibbitzFastener_PLA_1m42s.gcode                                    bowling rocker switch_PLA_2h6m.gcode                               vesa_nut.gcode

ProfessionalBawkBawkGames, 2025-05-07:
AD5M /usr/data # ls
octoapp-env

ProfessionalBawkBawkGames, 2025-05-07:
So, after restarting the printer and finally getting those envs to delete it seems we are now making progress

crysxd, 2025-05-07:
So /mnt/data is the print files it seems
Can you run `find / -name moonraker`? Might run a while. This will find where the other stuff is installed

crysxd, 2025-05-07:
Do this:
```
rm -rf /usr/data/octoapp
rm -rf /usr/data/octoapp-env
cd /root/
git clone https://github.com/crysxd/OctoApp-Plugin/ octoapp
git checkout k1-install-fix
./install.sh
```

crysxd, 2025-05-07:
Sorry, missed one command:

```
rm -rf /usr/data/octoapp
rm -rf /usr/data/octoapp-env
cd /root/
git clone https://github.com/crysxd/OctoApp-Plugin/ octoapp
cd octoapp
git checkout k1-install-fix
./install.sh
```

ProfessionalBawkBawkGames, 2025-05-07:
sooo not sure im meant to win this battle XD

Installing or updating required python libs...
ERROR: Exception:
Traceback (most recent call last):
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 438, in _error_catcher
    yield
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 561, in read
    data = self._fp_read(amt) if not fp_closed else b""
           ^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 527, in _fp_read
    return self._fp.read(amt) if amt is not None else self._fp.read()
           ^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 98, in read
    data: bytes = self.__fp.read(amt)
                  ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/http/client.py", line 473, in read
  File "/usr/lib/python3.11/socket.py", line 706, in readinto
  File "/usr/lib/python3.11/ssl.py", line 1314, in recv_into
  File "/usr/lib/python3.11/ssl.py", line 1166, in read
ssl.SSLError: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2580)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/cli/base_command.py", line 105, in _run_wrapper
    status = _inner_run()
             ^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/cli/base_command.py", line 96, in _inner_run
    return self.run(options, args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/cli/req_command.py", line 68, in wrapper
    return func(self, options, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/commands/install.py", line 387, in run
    requirement_set = resolver.resolve(
                      ^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 96, in resolve
    result = self._result = resolver.resolve(
                            ^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/resolvers/resolution.py", line 515, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/resolvers/resolution.py", line 388, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/resolvers/resolution.py", line 141, in _add_to_criteria
    if not criterion.candidates:
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/structs.py", line 194, in __bool__
    return bool(self._sequence)
           ^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 163, in __bool__
    self._bool = any(self)
                 ^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 147, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 37, in _iter_built
    candidate = func()
                ^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 187, in _make_candidate_from_link
    base: Optional[BaseCandidate] = self._make_base_candidate_from_link(
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 233, in _make_base_candidate_from_link
    self._link_candidate_cache[link] = LinkCandidate(
                                       ^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 306, in __init__
    super().__init__(
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 159, in __init__
    self.dist = self._prepare()
                ^^^^^^^^^^^^^^^
[... 12 lines cut ...]
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/operations/prepare.py", line 174, in unpack_url
    file = get_http_url(
           ^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/operations/prepare.py", line 115, in get_http_url
    from_path, content_type = download(link, temp_dir.path)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 560, in read
    with self._error_catcher():
  File "/usr/lib/python3.11/contextlib.py", line 158, in __exit__
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 449, in _error_catcher
    raise SSLError(e)
pip._vendor.urllib3.exceptions.SSLError: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2580)

crysxd, 2025-05-07:
These printer OS are horrible, it's not surprising
Run `date` and make sure it's correct

Then run this and see if it works
```
/root/printer_software/octoapp-env/bin/pip3 install --upgrade certifi
/root/printer_software/octoapp/install.sh
```

If not run this
```
mkdir -p /root/printer_software/octoapp-env/pip && echo -e "[global]\ntrusted-host = \n    pypi.org\n    files.pythonhosted.org\n    pypi.python.org" > /root/printer_software/octoapp-env/pip/pip.conf
/root/printer_software/octoapp/install.sh
```

ProfessionalBawkBawkGames, 2025-05-07:
Date was fine ran both same issue both times
`Installing or updating required python libs...
ERROR: Exception:
Traceback (most recent call last):
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 438, in _error_catcher
    yield
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 561, in read
    data = self._fp_read(amt) if not fp_closed else b""
           ^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 527, in _fp_read
    return self._fp.read(amt) if amt is not None else self._fp.read()
           ^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 98, in read
    data: bytes = self.__fp.read(amt)
                  ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/http/client.py", line 473, in read
  File "/usr/lib/python3.11/socket.py", line 706, in readinto
  File "/usr/lib/python3.11/ssl.py", line 1314, in recv_into
  File "/usr/lib/python3.11/ssl.py", line 1166, in read
ssl.SSLError: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2580)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/cli/base_command.py", line 105, in _run_wrapper
    status = _inner_run()
             ^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/cli/base_command.py", line 96, in _inner_run
    return self.run(options, args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/cli/req_command.py", line 68, in wrapper
    return func(self, options, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/commands/install.py", line 387, in run
    requirement_set = resolver.resolve(
                      ^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 96, in resolve
    result = self._result = resolver.resolve(
                            ^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/resolvers/resolution.py", line 515, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/resolvers/resolution.py", line 388, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/resolvers/resolution.py", line 141, in _add_to_criteria
    if not criterion.candidates:
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/resolvelib/structs.py", line 194, in __bool__
    return bool(self._sequence)
           ^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 163, in __bool__
    self._bool = any(self)
                 ^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 147, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 37, in _iter_built
    candidate = func()
                ^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 187, in _make_candidate_from_link
    base: Optional[BaseCandidate] = self._make_base_candidate_from_link(
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 233, in _make_base_candidate_from_link
    self._link_candidate_cache[link] = LinkCandidate(
                                       ^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 306, in __init__
    super().__init__(
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 159, in __init__
    self.dist = self._prepare()
                ^^^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 236, in _prepare
[... 11 lines cut ...]
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/operations/prepare.py", line 174, in unpack_url
    file = get_http_url(
           ^^^^^^^^^^^^^
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_internal/operations/prepare.py", line 115, in get_http_url
    from_path, content_type = download(link, temp_dir.path)
[... cut ...]
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 560, in read
    with self._error_catcher():
  File "/usr/lib/python3.11/contextlib.py", line 158, in __exit__
  File "/root/printer_software/octoapp-env/lib/python3.11/site-packages/pip/_vendor/urllib3/response.py", line 449, in _error_catcher
    raise SSLError(e)
pip._vendor.urllib3.exceptions.SSLError: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:2580)`

crysxd, 2025-05-08:
Was doing some more googling, this could be a networking issue. Could it be that the printer has a bad wifi connection? Devices like this can have very bad wifi hardware, so even when other devices have a good connection they can struggle.

Any chance you can put the printer next to your router and try again?
I also updated the install script to allow for more retries